OpenTracksApp/OpenTracks #2090: Fill out blanks in LICENSE file

residuum, 2025-04-15:
The license file is incomplete, and is missing date and copyright holder.

https://github.com/OpenTracksApp/OpenTracks/blob/ed163b2083038dce3459673fe50bf6d71b0cf78c/LICENSE#L189

This should be filled, especially if someone wants to take code and abide to the license.

dennisguse, 2025-04-15:
I need to check what I have to put there aka diving into the commit history...
But yeah, this should be done right.

dennisguse, 2025-04-16:
@residuum these blanks should not be filled in LICENSE.
Correct way to do this is the creation of a NOTICE file: https://infra.apache.org/licensing-howto.html#example-notice

dennisguse, 2025-04-24:
Actually, the NOTICE file is also optional (only required for Apache Foundation projecs).
So, no need to really do anything here.

From https://www.apache.org/licenses/LICENSE-2.0#apply (highlighting mine):

> Include a copy of the Apache License, typically in a file called LICENSE, in your work, and **consider** also including a NOTICE file that references the License.

residuum, 2025-04-24:
OK, so this is the preferred way to do it. It seems a bit strange, if someone wants to take the code and correctly reference the original authors and copyright date.

dennisguse, 2025-04-25:
It is actually complicated: until 2013(?) it was Google.
Then https://github.com/plonk42 did some cleaning.
In 2019, I did the actual fork to OpenTracks and maintain it since then.
However, a lot of people contribute since then.
So, the copyright holder since 2019 is not me alone (and there is no organization to reference).
No clue what to do in this case.

And why do you need this information?

residuum, 2025-04-25:
I am reading your code for implementing https://github.com/SensorApps/Sensors2OSC/issues/50 and probably will copy some portion of the architecture and data parsing logic from OpenTracks.

dennisguse, 2025-04-25:
@residuum have fun with the code as I have written this part (almost) completely. Just keep it under Apache and mention OpenTracks.