OpenTracksApp/OpenTracks #2092: Avoid import of already imported track does not work for exported Komoot-Tracks

update-freak, 2025-04-18:
**Describe the bug**
I imported a GPX-Track and later I imported the Track again.
Expectation was that the track is skipped because the setting was set to avoid double tracks.
The name of the file was not changed. I attached a screenshot and a file.
The file was exported from Komoot with a Python script.

[image]

[2025-02-21_Rundweg Kleingartach.gpx.txt](https://github.com/user-attachments/files/19816871/2025-02-21_Rundweg.Kleingartach.gpx.txt)

**To Reproduce**
1. Import the attached file twice

**Technical information**
 - Device: Samsung S23
 - OS: Android 14
 - OpenTracks version: 4.19

dennisguse, 2025-04-19:
When exporting OpenTracks adds a unique identifier into the track file (actually a UUID) and this value is later on used to determine if a track is already known or if it should be imported. 

Best guess: Komoot doesn't provide such an identifier. 

PS OpenTracks doesn't store the filename, so it cannot be used to recognize a file.

closed 2025-04-22 by dennisguse

update-freak, 2025-04-23:
ah ok.
Yes could be that there is no identifier from Komoot