OpenTracksApp/OpenTracks #2096: Export/Import file error for KMZ (with photos) files

itsmartashub, 2025-05-02:
**Describe the bug**
1. When exporting `KMZ` (**with photos**) files from OpenTracks, some files generate parser errors. When attempting to import these same files back into the app, the errors persist. This issue appears to be version-specific, as it doesn't occur in `v4.17.8`
2. Additionally, sharing these files to **Organic Maps** fails with a "failed to load bookmarks" error message. While Organic Maps doesn't support photo markers (expected behavior), KMZ files with photos should still load with the track data visible but without markers. Instead, these files fail completely, suggesting a file format issue rather than just the photo marker limitation.

**To Reproduce**
1. Export track data to KMZ (with photos) format
2. Attempt to import the KMZ (with photos) files back into OpenTracks
3. Alternatively, try sharing the exported KMZ (with photos) files to Organic Maps

**Error Messages**

1. When importing back to OpenTracks:
```
<file-name>.kmz: Parser error:
org.apache.harmony.xml.ExpatParser$ParseException: At line 1, column 0: no element found
```

2. When sharing to Organic Maps:
```
"Failed to load bookmarks. The file may be corrupted or defective"
```
<details>
<summary>When share KMZ to Organic Maps from OpenTracks crash log</summary>
### App information
* ID: de.dennisguse.opentracks.nightly
* Version: 6121 v4.19.1

### Device information
* Brand: Redmi
* Device: alioth
* Model: M2012K11AC
* Id: AQ3A.240912.001
* Product: alioth

### Firmware
* SDK: 35
* Release: 15
* Incremental: OS2.0.5.0.VNKCNXM

### Cause of error
```java
Exception in thread "AsyncTask #1": java.lang.RuntimeException: An error occurred while executing doInBackground()
	at android.os.AsyncTask$4.done(AsyncTask.java:415)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:381)
	at java.util.concurrent.FutureTask.setException(FutureTask.java:250)
	at java.util.concurrent.FutureTask.run(FutureTask.java:269)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1251)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:668)
	at java.lang.Thread.run(Thread.java:1012)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'float java.lang.Float.floatValue()' on a null object reference
	at de.dennisguse.opentracks.io.file.exporter.KMLTrackExporter.getHeading(KMLTrackExporter.java:478)
	at de.dennisguse.opentracks.io.file.exporter.KMLTrackExporter.writeMarker(KMLTrackExporter.java:277)
	at de.dennisguse.opentracks.io.file.exporter.KMLTrackExporter.writeMarkers(KMLTrackExporter.java:151)
	at de.dennisguse.opentracks.io.file.exporter.KMLTrackExporter.writeTrack(KMLTrackExporter.java:116)
	at de.dennisguse.opentracks.io.file.exporter.KMZTrackExporter.writeTrack(KMZTrackExporter.java:74)
	at de.dennisguse.opentracks.data.ShareContentProvider.lambda$openFile$3(ShareContentProvider.java:216)
	at de.dennisguse.opentracks.data.ShareContentProvider$$ExternalSyntheticLambda5.writeDataToPipe(D8$$SyntheticClass:0)
	at android.content.ContentProvider$1.doInBackground(ContentProvider.java:2572)
	at android.os.AsyncTask$3.call(AsyncTask.java:394)
	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
	... 3 more
```
</details>

**Screenshots**
[image]

[image]

[image]

**Technical information**
 - Device: Poco F3
 - OS: Android 15 | HyperOS `2.0.5.0` (custom ROM)
 - OpenTracks version: `v4.19.0` or `v4.19.1` Nightly (`6121`), all from fdroid
 - OpenTracks commit id (only for nightly builds): `6121`?

dennisguse, 2025-05-02:
Does this affect KML as well?

If not: which filename has the KML file inside the KMZ (It is just a zip file)?

itsmartashub, 2025-05-02:
Thanks for investigating this issue.

This only affects KMZ files with photo markers - standard KML, GPX, and KMZ without photos export correctly.

Looking at the problematic KMZ exports, it seems the photos directory is completely missing, containing only the `doc.kml` file inside the zip archive. In contrast, exports from `v4.17.8` properly include the photos directory for KMZ files with photo markers.

dennisguse, 2025-06-09:
@NebulaXS autsch.
Are you able to get a logcat?
I will add proper logging for problems in the export....

dennisguse, 2025-06-09:
Too fast ;)
The nightly is not yet built.
You can install -35 and then update to -36 (hopefully in the next hours).

dennisguse, 2025-06-10:
Good news: I put the log statement at the correct place.
Bad news: it only prints the same message.
I have added a `e.printStackTrace()` in addition (nooot nice).

PS/ One logcat is enough: both should show the same behavior.

dennisguse, 2025-06-10:
Thanks for diving in.
TBH I don't see the benefit of this code. 
It is ancient and not really doing something useful. 
I just kept it in there, because it was existing.

So, let's remove it and get the code base slightly nicer ;)

dennisguse, 2025-06-10:
@NebulaXS I didn't merge the PR yet. Let me do this now. The other update was a cleanup commit only.

commit 4843a238 (2025-06-10): Bugfix: KML/KMZ failed to export heading - ancient functionality got removed.

Fixes #2096.