OpenTracksApp/OpenTracks #2101: Cannot start OpenTracks using Gadgetbridge's debugging functionality

morags, 2025-05-17:
**Please describe the problem and how to reproduce it:**
Cannot start OpenTracks using Gadgetbridge's debugging functionality

Steps to reproduce:
1. Configure both apps as per [GB's wiki](https://gadgetbridge.org/internals/topics/opentracks-api/#how-to-use-this-practically).
2. On the GB debug menu, click "Fitness App Tracking Start"; GB logs the following:
```log
15:55:06.176 [main] WARN  n.f.g.u.GB - Unable to find explicit activity class {de.dennisguse.opentracks.playStore/de.dennisguse.opentracks.publicapi.StartRecording}; have you declared this activity in your AndroidManifest.xml?
```
3. On the GB debug menu, click "show Fit.App.Track. Status"; GB logs the following:
```log
15:55:10.168 [main] DEBUG n.f.g.a.DebugActivity - openTracksObserver debug watch dialog running
15:55:10.170 [main] DEBUG n.f.g.a.DebugActivity - openTracksObserver is null
```

Other notes:
OT tracks normally if started using OT itself.

**Checklist**
- [x] Consulted [https://dontkillmyapp.com/]
- [x] Power management: battery optimization is disabled for OpenTracks

**Technical information**

- Device: LG G6 (LG-H870)
- OS: Android 9
- OpenTracks version: 4.19.2
- Gadgetbridge version: 0.85.0

dennisguse, 2025-05-17:
Which version of OpenTracks did you install?
* reproducible via F-Droid or PlayStore
* no reproducible 
* nightly?

You have to configure this in Gadgetbridge as the three variants use different applicationIds.
PS/ In your case: the reproducible variant is configured in Gadgetbridge.

morags, 2025-05-17:
I installed the "reproducible build" through F-Droid. The app gives `de.dennisguse.opentracks.playStore` as the ID, and that's what I configured in GB.

dennisguse, 2025-05-17:
PublicAPI is enabled in OpenTracks?
See settings.

morags, 2025-05-17:
Public API is enabled, as is automatic data transfer.

dennisguse, 2025-05-17:
Can you try the ADB commands in the README?

It sounds as if it should be working...

morags, 2025-05-17:
It works:

```shell
> adb shell am start -n de.dennisguse.opentracks.playstore/de.dennisguse.opentracks.publicapi.StartRecording
Starting: Intent { cmp=de.dennisguse.opentracks.playstore/de.dennisguse.opentracks.publicapi.StartRecording }

> adb shell am start -n de.dennisguse.opentracks.playstore/de.dennisguse.opentracks.publicapi.StopRecording
Starting: Intent { cmp=de.dennisguse.opentracks.playstore/de.dennisguse.opentracks.publicapi.StopRecording }
```

dennisguse, 2025-05-17:
Can you check the applicationId that is used in Gadgetbridge. 

According to the error in your first message: it contains "playStore" (upcase s). 
In the adb commands it is lower case.

morags, 2025-05-17:
Well done - it works. I'll file this as a bug with Gadgetbridge - the dropdown setting should show the correct version.

Thanks!

closed 2025-05-17 by morags