OpenTracksApp/OpenTracks #2108: support themed icons

foss-, 2025-06-04:
**Describe the bug**
Currently with themed icons enabled on Android the icons for OpenTracks (reproducible and non-reproducible) are showing non-themed icons. Assuming themed icons are not yet supported.
Using Kvaesitso launcher and themed icons, the app icons show as themed icons but the icon shows too small:

[image]

**To Reproduce**
1. install kvaesitso launcher
2. enable themed icons
or
1. enable themed icons in default launcher and see non themed icons

**Technical information**
 - Device: Xiaomi lisa
 - OS: ArrowOS 13.1
 - OpenTracks version: 4.19.3

foss-, 2025-06-04:
This is the [themed app icon](https://developer.android.com/develop/ui/views/launch/icon_design_adaptive) documentation by Google.

dennisguse, 2025-06-05:
@foss- actually, why do you say the themed icon is not working correctly?

foss-, 2025-06-05:
The icon is much smaller than the rest of the themed icons. I'd expect the themed icon to use the same size that arrow icon uses in non-themed view.

foss-, 2025-06-06:
[Kvaesitso](https://github.com/MM2-0/Kvaesitso) could indeed be the issue. But OpenTracks is the only app I am seeing this issue with. Oh and I just double checked on an Android 15 device and the non themed icon exposes the same problematic behavior there.

[image]

foss-, 2025-06-21:
Installed 4.19.5 via Obtanium and sadly the issue is persisting.

dennisguse, 2025-06-21:
@foss- damn. I forgot the icon for the reproducible build (only for nightly and debug). 
Will do this change later today. 
Thanks for testing and reporting!

foss-, 2025-06-23:
Installed 4.19.6 and restarted the device for good measure. Sadly the icon still shows in small size.

foss-, 2025-06-28:
Confirming that 4.20.0 indeed resolves the issue. A restart was required for the changed to be picked up by kvaesitso. Thanks again for the fix.

PR #2116 (2025-06-14): Launcher icon was sometimes incorrectly scaled (e.g., kvaesitso).
Use same icon as OSMDashboard for launcher (w/o background).

Fixes #2108.

@pstorch I just took your approach using separate drawables.
Not an issue to think longer about - thanks for preparing the solution so nicely :D
commit 77d040d0 (2025-06-14): Launcher icon was sometimes incorrectly scaled (e.g., kvaesitso).
Use same icon as OSMDashboard for launcher (w/o background).

Fixes #2108.
commit 3bae0869 (2025-06-21): Launcher icon was sometimes incorrectly scaled (e.g., kvaesitso).
Use same icon as OSMDashboard for launcher (w/o background).

This time also for the reproducible build.

Fixes #2108.
commit e41df739 (2025-06-29): Launcher icon was sometimes incorrectly scaled (e.g., kvaesitso).
Use same icon as OSMDashboard for launcher (w/o background).

This time, we re-add the proper icon (white) for the reproducible builds.

Fixes #2108.