OpenTracksApp/OpenTracks #2126: Add a way to launch the stats dashboard from public API

kflpk, 2025-06-29:
**Is your feature request related to a problem? Please describe.**
I'm trying to use Tasker to automatically start the recording when using an NFC tag.
Currently I don't think there is a way to launch the statistics dashboard (not the map) from the public API.

**Describe the solution you'd like**
A new public API function function to launch the dashboard for currently recorded track, or a parameter to `StartRecording` to do the same thing. 

**Describe alternatives you've considered**
Currently I only have Tasker launch the app, but it kinda defeats the purpose, as I have to click on the screen anyway.

**Additional context**

https://github.com/user-attachments/assets/693880a2-290e-4abe-acbf-ed362434be1b

dennisguse, 2025-06-29:
Can you do the change yourself?
It is in StartRecording and should just be a call to startActivity(TrackRecordingActivity).

kflpk, 2025-06-29:
I could try, BUT I don' know anything about programming in Java and Android development. I barely got the intent working, as I am not familiar with the project's codebase nor Android APIs.

dennisguse, 2025-06-29:
Then one last question: OpenTracks starts the UI if triggered via the PublicAPI or am I mistaken?