OpenTracksApp/OpenTracks #2135: becoming idle while user keeps moving

foss-, 2025-07-10:
**Describe the bug**
During a run yesterday I encountered the audio info `becoming idle` multiple times despite being on the move. In the end idle time accumulated to a total of six minutes. Since I know I did not take a break this information is incorrect. Ideally Open Tracks would not go into idle mode when user is still moving.

Trees may have been involved as the run was not under clear sky.

**Technical information**
- Device: Xiaomi lisa
- OS: Android 13.1, ArrowOS
 - OpenTracks version: 4.21.0

dennisguse, 2025-07-12:
Please check settings: Recording > Idle Threshold.
If the reception is bad, 10s may not be enough.

closed 2025-07-12 by dennisguse

foss-, 2025-07-12:
Thanks, upped to 15s. Will test and may up to 30 if needed.