OpenTracksApp/OpenTracks #2140: Automatic GPX export to Gadgetbridge via public API

arjan-s, 2025-07-15:
**Is your feature request related to a problem? Please describe.**
Gadgetbridge currently uses the public API to be able to start and stop tracking and to receive live statistics while tracking. I'd like a way to automatically import GPX files into Gadgetbridge as soon as the tracking is stopped (by Gadgetbridge).

**Describe the solution you'd like**
An API method of extracting the GPX data from the just-stopped track.

**Describe alternatives you've considered**
Manually (or automatically) exporting GPX files from OpenTracks and then manually importing them into Gadgetbridge. This is time intensive, which is what I would like to solve.
I also know about the TRACKPOINTS_URI_INDEX Uri that's being shared with Gadgetbridge upon starting tracking, but it is unclear to me whether that should/can be used for this purpose.

**Additional context**
I'm a Gadgetbridge developer and able to change things on our side if required.

arjan-s, 2025-07-16:
Thanks! Both options are valid for this use case.
What I want is to let Gadgetbridge retrieve (or generate) a GPX file at the end of a recording that was also started (and will be stopped) by Gadgetbridge. So the mentioned limitation is fine.

Gadgetbridge requires an actual GPX file for displaying the recorded track in a workout summary. So rather than generating a GPX file myself (from the received track points) I think I'll look into the extra URI possibility, as OpenTracks already has GPX generation code.

arjan-s, 2025-07-16:
Well, I may have typed that before actually looking at the code... :)
No, the rendering indeed doesn't require GPX, it's actually our own code that assumes it.

In the mean time I already implemented GPX track saving using the existing trackpoints URI together with the GPX file generator that we already had in the Gadgetbridge code. So I won't be having to change OpenTracks.

You seem to have something against GPX, would you care to enlighten me? :) I don't know much about this subject, so if there is a better way or a better file format, I'd like to know!

arjan-s, 2025-07-16:
Ahh, so you'd prefer a more flexible format, I get that.
For my current use case these limitations are perfect though. :)

dennisguse, 2025-07-17:
Trackpoint.getLatitude() /get longitude() may be null.

TrackPoint.getTime() should be present (instead of new Date()). Is never null.

Besides that looks good :)

PS just have a phone at hand...

dennisguse, 2025-07-19:
Internally OpenTracks may be null lat/lng, so I would account for it. Otherwise you may see the famous 0,0 location (West of Africa). 

TBH I have not yet worked often with the OSMDashoard code; I only know the OpenTracks side in detail.

arjan-s, 2025-07-20:
I thought so, and I'm already accounting for that in my modifications of the TrackPoint class. So I'm pretty sure it'll be fine! In any case I've tested it for a few days myself already, and I'm very happy with the results. :)

arjan-s, 2025-07-20:
For reference, I've made a PR here: https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/5121

arjan-s, 2025-07-24:
Ah this is awesome, thanks again! Yes, elevation is already in there, so I could simply use it!
I'll close this issue now, as OpenTracks already supports everything I hoped for. :)

closed 2025-07-24 by arjan-s